sandeepmistry/node-sx127x #1: About npm problem

fxsh1301, 2017-05-16:
Hi sir ! Thanks for sharing the code for Raspberry Pi.
I met some problem when i was installing node-sx127x
Here's the steps I did and the error message :
`pi@raspberrypi:~ $ sudo npm install sx127x
pi@1.0.0 /home/pi
└── sx127x@0.0.0

npm WARN pi@1.0.0 No description
npm WARN pi@1.0.0 No repository field.
pi@raspberrypi:~ $ cd sx127x/node-sx127x-master/
pi@raspberrypi:~/sx127x/node-sx127x-master $ ls
API.md  examples  index.js  lib  LICENSE  npm-debug.log  package.json  README.md
pi@raspberrypi:~/sx127x/node-sx127x-master $ cd examples/
pi@raspberrypi:~/sx127x/node-sx127x-master/examples $ ls
npm-debug.log  receiver.js  sender.js
pi@raspberrypi:~/sx127x/node-sx127x-master/examples $ node receiver.js
open { Error: ENOENT, No such file or directory
    at Error (native) errno: 2, code: 'ENOENT', syscall: 'open' }
/home/pi/sx127x/node-sx127x-master/examples/receiver.js:14
    throw err;
    ^

Error: ENOENT, No such file or directory
    at Error (native)
`

sandeepmistry, 2017-05-29:
@fxsh1301 try running the `npm install` in a directory not names `sx127x`.

closed 2017-05-29 by sandeepmistry

fxsh1301, 2017-06-05:
https://www.hackster.io/idreams/getting-started-with-lora-fd69d1

[image: 內置圖片 1]
因為柯大您再露天上面有提到和Dragino相容,所以我用這篇去試
我確認我的SPI介面是有開啟的,因為拿Arduino透過SPI接Pi是有讀到的
所以不確定是不是miniLora的問題,我使用的是Pi3,但不同版本的Pi其SPI腳位都相同

Yi-Siang Ciou, B.Eng.

Taiwan, R.O.C.

E-mail: fxsh13002@gmail.com

Line ID :fxsh1301