sandeepmistry/node-sx127x #2: Can not install via npm command

fxsh1301, 2017-06-05:
Hi sir! I still can not install node-sx127x on my Pi3.
My npm's version is 1.4.21 . And it showed many errors after I use " npm install sx127x".
Here's the screenshot of my Pi3. Or is there any other kit that I have to install before sx127x?
Thanks for reply!
[http://imgur.com/Ifq8WCC](url)
[http://imgur.com/DiCpQYr](url)

sandeepmistry, 2017-06-18:
@fxsh1301 how did you install things?

I suggest the following:
```
cd ~
mkdir test-node-sx127x
cd  test-node-sx127x
npm install sx127x
cd node_modules/sx127x/examples
sudo node sender.js
```

closed 2017-06-18 by sandeepmistry

fxsh1301, 2017-07-04:
@sandeepmistry 
Hi sir, I tried to install with you said but met some problem.

pi@raspberrypi:~ $ cd ~
pi@raspberrypi:~ $ mkdir test-node-sx127x
pi@raspberrypi:~ $ cd  test-node-sx127x
pi@raspberrypi:~/test-node-sx127x $ npm install sx127x

> epoll@0.1.22 install /home/pi/test-node-sx127x/node_modules/sx127x/node_modules/onoff/node_modules/epoll
> node-gyp rebuild

make: Entering directory '/home/pi/test-node-sx127x/node_modules/sx127x/node_modules/onoff/node_modules/epoll/build'
  CXX(target) Release/obj.target/epoll/src/epoll.o
In file included from ../src/epoll.cc:15:0:
../node_modules/nan/nan.h:326:47: error: ‘REPLACE_INVALID_UTF8’ is not a member of ‘v8::String’
   static const unsigned kReplaceInvalidUtf8 = v8::String::REPLACE_INVALID_UTF8;
                                               ^
epoll.target.mk:84: recipe for target 'Release/obj.target/epoll/src/epoll.o' failed
make: *** [Release/obj.target/epoll/src/epoll.o] Error 1
make: Leaving directory '/home/pi/test-node-sx127x/node_modules/sx127x/node_modules/onoff/node_modules/epoll/build'
gyp ERR! build error
gyp ERR! stack Error: `make` failed with exit code: 2
gyp ERR! stack     at ChildProcess.onExit (/usr/share/node-gyp/lib/build.js:267:23)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.9.35-v7+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/test-node-sx127x/node_modules/sx127x/node_modules/onoff/node_modules/epoll
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian

npm ERR! epoll@0.1.22 install: `node-gyp rebuild`
npm ERR! Exit status 1
npm ERR!
npm ERR! Failed at the epoll@0.1.22 install script.
npm ERR! This is most likely a problem with the epoll package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     node-gyp rebuild
npm ERR! You can get their info via:
npm ERR!     npm owner ls epoll
npm ERR! There is likely additional logging output above.

npm ERR! System Linux 4.9.35-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "sx127x"
npm ERR! cwd /home/pi/test-node-sx127x
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm ERR! Error: ENOENT, lstat '/home/pi/test-node-sx127x/node_modules/sx127x/node_modules/async/node_modules/lodash/lodash.js'
npm ERR! If you need help, you may report this *entire* log,
npm ERR! including the npm and node versions, at:
npm ERR!     <http://github.com/npm/npm/issues>

npm ERR! System Linux 4.9.35-v7+
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "install" "sx127x"
npm ERR! cwd /home/pi/test-node-sx127x
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! path /home/pi/test-node-sx127x/node_modules/sx127x/node_modules/async/node_modules/lodash/lodash.js
npm ERR! fstream_path /home/pi/test-node-sx127x/node_modules/sx127x/node_modules/async/node_modules/lodash/lodash.js
npm ERR! fstream_type File
npm ERR! fstream_class FileWriter
npm ERR! code ENOENT
npm ERR! errno 34
npm ERR! fstream_stack /usr/lib/nodejs/fstream/lib/writer.js:284:26
npm ERR! fstream_stack Object.oncomplete (evalmachine.<anonymous>:107:15)

> spi-device@0.2.6 install /home/pi/test-node-sx127x/node_modules/sx127x/node_modules/spi-device
> node-gyp rebuild

gyp: binding.gyp not found (cwd: /home/pi/test-node-sx127x/node_modules/sx127x/node_modules/spi-device) while trying to load binding.gyp
gyp ERR! configure error
gyp ERR! stack Error: `gyp` failed with exit code: 1
gyp ERR! stack     at ChildProcess.onCpExit (/usr/share/node-gyp/lib/configure.js:344:16)
gyp ERR! stack     at ChildProcess.emit (events.js:98:17)
gyp ERR! stack     at Process.ChildProcess._handle.onexit (child_process.js:809:12)
gyp ERR! System Linux 4.9.35-v7+
gyp ERR! command "nodejs" "/usr/bin/node-gyp" "rebuild"
gyp ERR! cwd /home/pi/test-node-sx127x/node_modules/sx127x/node_modules/spi-device
gyp ERR! node -v v0.10.29
gyp ERR! node-gyp -v v0.12.2
gyp ERR! not ok
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian

npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /home/pi/test-node-sx127x/npm-debug.log
npm ERR! not ok code 0
pi@raspberrypi:~/test-node-sx127x $

fxsh1301, 2017-07-05:
Hi sir.
It worked after I update the npm.
But I found Arduino can't send data to Pi.
However the Pi can send data to Arduino.
What kind of problem it might be ?
Thanks!

Yi-Siang Ciou, B.Eng.

Taiwan, R.O.C.

E-mail: fxsh13002@gmail.com

Line ID :fxsh1301

fxsh1301, 2017-07-27:
Hi sir.
I still can't figure out the problem.
Will you mind to check the issue that I post on node-sx127x?
The code work great with Arduino but the receiver.js on Pi got some problem.
Thanks!

Yi-Siang Ciou, B.Eng.

Taiwan, R.O.C.

E-mail: fxsh13002@gmail.com

Line ID :fxsh1301

2017-07-05 11:24 GMT+08:00 Yi-Siang Ciou <fxsh13002@gmail.com>: