sandeepmistry/node-sx127x #3: Request addition of close method that uses this._spi.closeSync

solutionsbyconnections, 2017-06-28:
While wrapping node-sx127x in a node-red node I found that when node-red sends a "close" to my node during "deploy" or exit from node-red the _spi.close does not complete in time to return the callback from a call from 'close'.  Upon restarting node-red the sx127x can not be opened without rebooting the orangepi zero (running Armbain's version of Ubuntu 16.04).

I found a workaround by adding a new method, closeSync that uses the spi-device 'closeSync' which requires no call back.  Here is the code:

SX127x.prototype.closeSync = function() {
    console.log("got a closeSync of sx127x");
    this._spi.closeSync();
    this._spi = null;
    this._dio0Gpio.unexport();
    this._resetGpio.unexport();
};

Thanks for the great code,
- Bob

fxsh1301, 2017-07-05:
@solutionsbyconnections 
Hi sir!
Sorry for asking about other question.
Did you success transmit two-way between Arduino and Raspberry Pi?
I got some issue here, can't send data to Pi from Arduino.

solutionsbyconnections, 2017-07-07:
Hi,
I did the sx127x to work with Arduino RadioTime using the reliable datagram service.  I am in the process of rebuilding my development environment over the next few days.  Once I am finished I can summarize the changes I needed to make to the SX127x code as well as the Node code for sending and receiving reliable datagrams from RadioTime using the RFM95 driver/manager.

sandeepmistry, 2017-09-05:
@solutionsbyconnections could you please submit a pull request for this change. Thanks.