sandeepmistry/node-sx127x #5: open Error: Invalid version 2, expected 0x12

skoky, 2017-12-07:
Please advice why example does not work on my Raspi / Dragino board 

```
pi@raspberrypi:~/node-sx127x/examples $ nodejs sender.js
open Error: Invalid version 2, expected 0x12
    at SX127x.<anonymous> (/home/pi/node-sx127x/lib/sx127x.js:124:23)
    at /home/pi/node-sx127x/node_modules/async/dist/async.js:3874:9
    at /home/pi/node-sx127x/node_modules/async/dist/async.js:473:16
    at replenish (/home/pi/node-sx127x/node_modules/async/dist/async.js:993:25)
    at iterateeCallback (/home/pi/node-sx127x/node_modules/async/dist/async.js:983:17)
    at /home/pi/node-sx127x/node_modules/async/dist/async.js:958:16
    at /home/pi/node-sx127x/node_modules/async/dist/async.js:3871:13
    at SX127x.<anonymous> (/home/pi/node-sx127x/lib/sx127x.js:472:5)
/home/pi/node-sx127x/examples/sender.js:14
    throw err;
    ^

Error: Invalid version 2, expected 0x12
    at SX127x.<anonymous> (/home/pi/node-sx127x/lib/sx127x.js:124:23)
    at /home/pi/node-sx127x/node_modules/async/dist/async.js:3874:9
    at /home/pi/node-sx127x/node_modules/async/dist/async.js:473:16
    at replenish (/home/pi/node-sx127x/node_modules/async/dist/async.js:993:25)
    at iterateeCallback (/home/pi/node-sx127x/node_modules/async/dist/async.js:983:17)
    at /home/pi/node-sx127x/node_modules/async/dist/async.js:958:16
    at /home/pi/node-sx127x/node_modules/async/dist/async.js:3871:13
    at SX127x.<anonymous> (/home/pi/node-sx127x/lib/sx127x.js:472:5)```

sandeepmistry, 2017-12-18:
@skoky what chip is your Dragino board based on? Could you share a photo of your setup and explain the wires you connected?

Lyliya, 2017-12-18:
ive got same error with dragino v1.4

`open Error: Invalid version 192, expected 0x12
    at SX127x.<anonymous> (/home/pi/node_lora/lora/node_modules/sx127x/lib/sx127x.js:120:23)
    at /home/pi/node_lora/lora/node_modules/async/dist/async.js:3874:9
    at /home/pi/node_lora/lora/node_modules/async/dist/async.js:473:16
    at replenish (/home/pi/node_lora/lora/node_modules/async/dist/async.js:993:25)
    at iterateeCallback (/home/pi/node_lora/lora/node_modules/async/dist/async.js:983:17)
    at /home/pi/node_lora/lora/node_modules/async/dist/async.js:958:16
    at /home/pi/node_lora/lora/node_modules/async/dist/async.js:3871:13
    at SX127x.<anonymous> (/home/pi/node_lora/lora/node_modules/sx127x/lib/sx127x.js:468:5)
/home/pi/node_lora/lora/index.js:14
    throw err;
    ^

Error: Invalid version 192, expected 0x12
    at SX127x.<anonymous> (/home/pi/node_lora/lora/node_modules/sx127x/lib/sx127x.js:120:23)
    at /home/pi/node_lora/lora/node_modules/async/dist/async.js:3874:9
    at /home/pi/node_lora/lora/node_modules/async/dist/async.js:473:16
    at replenish (/home/pi/node_lora/lora/node_modules/async/dist/async.js:993:25)
    at iterateeCallback (/home/pi/node_lora/lora/node_modules/async/dist/async.js:983:17)
    at /home/pi/node_lora/lora/node_modules/async/dist/async.js:958:16
    at /home/pi/node_lora/lora/node_modules/async/dist/async.js:3871:13
    at SX127x.<anonymous> (/home/pi/node_lora/lora/node_modules/sx127x/lib/sx127x.js:468:5)
`
[image]

rgillan, 2017-12-27:
Hi,
was having trouble also with my v1.3.  You will see that the Dragino hat wires NSS (SPI CE) out to pin 22 (a GPIO) and not to either CE0 (24) or CE1 (26).  If you solder a jumper wire between pins 22 and 24 on the hat it should solve the issue I expect.  Further info in:
https://www.hackster.io/idreams/getting-started-with-lora-fd69d1

vynci, 2018-01-07:
hi is this board supported? 

http://wiki.dragino.com/index.php?title=Lora_Shield

i'm getting this error:

open Error: Invalid version 0, expected 0x12
    at SX127x.<anonymous> (/home/pi/first-ocean/backend/node_modules/sx127x/lib/sx127x.js:120:23)
    at /home/pi/first-ocean/backend/node_modules/sx127x/node_modules/async/dist/async.js:3874:9
    at /home/pi/first-ocean/backend/node_modules/sx127x/node_modules/async/dist/async.js:473:16
    at replenish (/home/pi/first-ocean/backend/node_modules/sx127x/node_modules/async/dist/async.js:993:25)
    at iterateeCallback (/home/pi/first-ocean/backend/node_modules/sx127x/node_modules/async/dist/async.js:983:17)
    at /home/pi/first-ocean/backend/node_modules/sx127x/node_modules/async/dist/async.js:958:16
    at /home/pi/first-ocean/backend/node_modules/sx127x/node_modules/async/dist/async.js:3871:13
    at SX127x.<anonymous> (/home/pi/first-ocean/backend/node_modules/sx127x/lib/sx127x.js:468:5)
/home/pi/first-ocean/backend/lora.js:14
    throw err;
    ^

Error: Invalid version 0, expected 0x12
    at SX127x.<anonymous> (/home/pi/first-ocean/backend/node_modules/sx127x/lib/sx127x.js:120:23)
    at /home/pi/first-ocean/backend/node_modules/sx127x/node_modules/async/dist/async.js:3874:9
    at /home/pi/first-ocean/backend/node_modules/sx127x/node_modules/async/dist/async.js:473:16
    at replenish (/home/pi/first-ocean/backend/node_modules/sx127x/node_modules/async/dist/async.js:993:25)
    at iterateeCallback (/home/pi/first-ocean/backend/node_modules/sx127x/node_modules/async/dist/async.js:983:17)
    at /home/pi/first-ocean/backend/node_modules/sx127x/node_modules/async/dist/async.js:958:16
    at /home/pi/first-ocean/backend/node_modules/sx127x/node_modules/async/dist/async.js:3871:13
    at SX127x.<anonymous> (/home/pi/first-ocean/backend/node_modules/sx127x/lib/sx127x.js:468:5)

thanks

vynci, 2018-01-07:
nevermind, i made it to work my lowering the SPI frequency from 12E6 to 8E6

var SPI_OPTIONS = {
  mode: spi.MODE0,
  maxSpeedHz: 8E6
};

also tried 4E6 and 1E6, they seem to be working

sandeepmistry, 2018-01-08:
Has anyone tried using the options to pass in the pins the board uses: https://github.com/sandeepmistry/node-sx127x/blob/master/API.md#create-device

epasinetti, 2020-12-18:
> nevermind, i made it to work my lowering the SPI frequency from 12E6 to 10E6
> 
> var SPI_OPTIONS = {
> mode: spi.MODE0,
> maxSpeedHz: 10E6
> };

I have modified these options but not solved the problem, have you got some suggestions ?