sandeepmistry/node-sx127x #11: Invalid version 0

sekisek, 2018-04-05:
Hi have Raspberry Pi 3b.

I am trying to connect sx1278.

[The radio module](https://www.ebay.com/itm/SX1278-433MHz-Wireless-Transceiver-Module-LoRa-SPSP-Long-Distance/332012370467?_trkparms=aid%3D111001%26algo%3DREC.SEED%26ao%3D1%26asc%3D50425%26meid%3D6a436e1489c74837869d683ec8e1313a%26pid%3D100675%26rk%3D1%26rkt%3D15%26mehot%3Dag%26sd%3D332012370467%26itm%3D332012370467&_trksid=p2481888.c100675.m4236&_trkparms=pageci%3A540f87c5-36be-11e8-9646-74dbd180ddbd%7Cparentrq%3A884d049f1620a9c9a6029f34ffffde7f%7Ciid%3A1)

I get the same error for the sender and reciver

node sender.js 
open Error: Invalid version 0, expected 0x12
    at SX127x.<anonymous> (/home/pi/node-sx127x/lib/sx127x.js:124:23)
    at /home/pi/node_modules/async/dist/async.js:3874:9
    at /home/pi/node_modules/async/dist/async.js:473:16
    at replenish (/home/pi/node_modules/async/dist/async.js:993:25)
    at iterateeCallback (/home/pi/node_modules/async/dist/async.js:983:17)
    at /home/pi/node_modules/async/dist/async.js:958:16
    at /home/pi/node_modules/async/dist/async.js:3871:13
    at SX127x.<anonymous> (/home/pi/node-sx127x/lib/sx127x.js:472:5)
/home/pi/node-sx127x/examples/sender.js:14

any Idea?

lakewik, 2018-04-15:
Hi
Module probably isn't connected properly or you don't enabled SPI interface in Raspberry Pi configuration (sudo raspi-config).

aylashiv, 2018-04-15:
Also provide sufficient power to the LoRa Module as it draws good power, Better to provide external power supply 3.3V rather than feeding it from the raspberrypi's 3.3v.

lorenzoromagnoli, 2019-04-15:
Hey @sandeepmistry! 

I'm also experiencing the same issue using the[ Dragino lora pi hat v 1.4](https://wiki.dragino.com/index.php?title=Lora/GPS_HAT) with a raspberry pi 3

I also tried modifying the pins used inside the library but had no better luck. 

I managed to have the communication between the pi and the lora module working using this script https://github.com/tftelkamp/single_chan_pkt_fwd, So I don't think there is a power or a configuration issue. 

Any other suggestion on how to solve this?

epasinetti, 2020-12-18:
> Hey @sandeepmistry!
>
> I'm also experiencing the same issue using the[ Dragino lora pi hat v 1.4](https://wiki.dragino.com/index.php?title=Lora/GPS_HAT) with a raspberry pi 3
>
> I also tried modifying [...]

Same problem here ...have you solved ?